pjueon/JetsonGPIO #1: GPIO input issue

ngminhtritm, 2019-11-16:
Hi guy,
I found that an issue from your code, you don't return value in GPIO::input(const string& channel), so this function has the random value.
Please review it.
BR

@@ -557,7 +560,7 @@ int GPIO::input(const string& channel){
 }
 
 int GPIO::input(int channel){
-    input(to_string(channel));
+    return input(to_string(channel));
 }

yomuevans, 2019-12-10:
hi cant compile for some reason. Here's the output to "g++ -o simple_out.cpp -lJetsonGPIO"

/usr/lib/gcc/aarch64-linux-gnu/7/../../../aarch64-linux-gnu/Scrt1.o: In function `_start':
(.text+0x18): undefined reference to `main'
(.text+0x1c): undefined reference to `main'
collect2: error: ld returned 1 exit status

lg93lggt, 2020-01-17:
I have an issue, my input retvel = -1855670792,  but when I use python gpio it runs normally.

closed 2021-04-18 by pjueon