pjueon/JetsonGPIO #4: Input Issue

lg93lggt, 2020-01-17:
I have an issue, my input retvel = -1855670792, but when I use python gpio it runs normally.

sunfic, 2020-05-28:
I have similar issue. there is always big negative value

rose-jinyang, 2020-08-10:
Hi
I have same issue.
Is there any solution?

svins, 2020-09-20:
There is a bug. A return statement is missing from this line:
https://github.com/pjueon/JetsonGPIO/blob/0d305e0b149151de61f336d0d8a8ddf4725be7da/src/JetsonGPIO.cpp#L560

MeijisIrlnd, 2020-09-30:
> 
> 
> There is a bug. A return statement is missing from this line:
> 
> https://github.com/pjueon/JetsonGPIO/blob/0d305e0b149151de61f336d0d8a8ddf4725be7da/src/JetsonGPIO.cpp#L560

yooo you just saved me hours of frustration thanks so much man

closed 2021-04-18 by pjueon