pjueon/JetsonGPIO #13: PWM not working

alexisguiter, 2021-01-02:
Hi everyone,

I'm trying to  get PWM to work on my Xavier NX. I modified the code in the manner described here to get the NX to work with this library: https://github.com/pjueon/JetsonGPIO/pull/10

I am able to get inputs/outputs to work,

however, when testing the PWM sample code, I get this issue: 

terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Aborted (core dumped)

Here's the sample code I used:

`
#include <iostream>
// for delay function.
#include <chrono>
#include <thread>
#include <map>
#include <string>

// for signal handling
#include <signal.h>

#include <JetsonGPIO.h>

using namespace std;

const map<string, int> output_pins{{"JETSON_XAVIER", 18}, {"JETSON_NANO", 33}, {"JETSON_NX", 33}};

//if(output_pins.find(GPIO::model) == output_pins.end() ){
//      cerr << "PWM not supported on this board\n";
//      terminate();
//}

// Pin Definitions
const int output_pin = output_pins.at(GPIO::model);

bool end_this_program = false;

inline void delay(int s){
        this_thread::sleep_for(chrono::seconds(s));
}

void signalHandler (int s){
        end_this_program = true;
}

int main(){

if(output_pins.find(GPIO::model) == output_pins.end() ){
        cerr << "PWM not supported on this board\n";
        terminate();
}

        // When CTRL+C pressed, signalHandler will be called
        signal(SIGINT, signalHandler);

        // Pin Setup.
        // Board pin-numbering scheme
        GPIO::setmode(GPIO::BOARD);

        // set pin as an output pin with optional initial state of HIGH
        GPIO::setup(output_pin, GPIO::OUT);
        GPIO::PWM p(output_pin, 50);
        p.start(5);

        cout << "PWM running. Press CTRL+C to exit." << endl;

        while(!end_this_program){
                delay(1);
        }

        p.stop();
        GPIO::cleanup();

        return 0;
}
`

rogerartigaspursals, 2021-01-12:
Hi,

I'm having a similar problem. The library behaves ok for simple read/write, but it's not working for PWM. I did everything described in the installation and tried on the Nano, AGX, and NX. The message is allways "Permission denied: path: /pwm0/enable". Even with sudo it seems that there is a permission issue. The hardware is working and the PWM can be activated and modified using python, but not in C++ due to this permission message.

Does anyone know what can be the problem and a solution for it?

Many thanks to all

Roger

JKI757, 2021-01-29:
https://github.com/JKI757/JetsonGPIO

closed 2021-04-24 by pjueon