pjueon/JetsonGPIO #17: Docker container

sashabel, 2021-04-22:
Hello, I've a question.
How can I use the library from within docker container ?
I'm using Jetson nano

sashabel, 2021-04-22:
My Dockerfile:
```
FROM ubuntu:20.04

ARG uid
ARG gid
ARG gid_gpio
RUN groupadd -f -r -g $gid_gpio gpio
RUN groupadd -f -r -g 1000 user

# gpio needs to be main group, otherwise we get permission problems
RUN useradd -M --uid $uid -g $gid_gpio user --groups uucp,$gid,user 
RUN mkdir -p /etc/sudoers.d \ 
   && echo 'user ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers.d/user \
   && echo 'Defaults exempt_group+=user' >> /etc/sudoers.d/user \
   && chmod a=r,o= /etc/sudoers.d/user

WORKDIR /app
COPY simple_out /app
RUN chmod +x simple_out

CMD ["/bin/sh", "-c", "/app/simple_out"]

```

Run command:
```
docker run --rm -v /sys/class/gpio:/sys/class/gpio -v /sys/class/pwm:/sys/class/pwm --device /dev/spidev0.0:/dev/spidev0.0:rw --group-add $(cut -d: -f3 < <(getent group gpio)) "my_image_name"
```

Error output:
```
[Exception] Could not determine Jetson model (catched from: get_data())
terminate called after throwing an instance of 'bool'
Aborted (core dumped)

```

pjueon, 2021-04-25:
The exception you got is from get_data() function in src/gpio_pin_data.cpp
The library determines which Jetson model you're using by reading /proc/device-tree/compatible file
I'm not familiar with docker so I'm not sure, but can you access to /proc/device-tree/compatible file from your environment?

And can you use NVIDIA's official Jetson GPIO Python library ?
The official python library also determines your Jetson model in the same way. 

If you can't use the official python library too, you might ask some advice to the official github  reposiory .(https://github.com/NVIDIA/jetson-gpio)

Unfortunately, I don't have any Jetson device to test right now so I can't offer you much information.

pjueon, 2021-05-25:
https://github.com/NVIDIA/jetson-gpio/issues/52

closed 2021-05-25 by pjueon