pjueon/JetsonGPIO #20: PWM - GPIO::PWM p(output_pin, 50);

TungTT1311, 2021-05-28:
I declare "p" in one function, and use it in another. 

How can I declare "p" as a global variable?

pjueon, 2021-05-30:
I'm not sure if I understand your question correctly. Maybe something like this? 
```cpp
// factory function for PWM
GPIO::PWM CreatePWMObject(int output_pin)
{
        // Pin Setup.	
        // Board pin-numbering scheme	
        GPIO::setmode(GPIO::BOARD);
	// set pin as an output pin with optional initial state of HIGH	
        GPIO::setup(output_pin, GPIO::OUT, GPIO::HIGH);	

        GPIO::PWM p(output_pin, 50);<!--EndFragment-->

        return p;
}

// declare "p" as a global variable
static GPIO::PWM p = CreatePWMObject(output_pin, 50);

int main()
{
    // your code...
}
```

TungTT1311, 2021-06-01:
Oke thanks you

TungTT1311, 2021-06-01:
/home/jetson/Desktop/person_detection/pedestrian_detection/test.cpp:246:10: error: use of deleted function ‘GPIO::PWM::PWM(const GPIO::PWM&)’
   return p;
I used it but error.

pjueon, 2021-06-01:
Oops, my bad. I forgot that PWM objects cannot be copied. 
I added move constructor/move assign operator to PWM class just now.

In JetsonGPIO folder, update your local repository.  
```git pull``` 

Rebuild & reinstall.
```
cd build
make all
sudo make install
```

And try the above example code again.

closed 2021-06-07 by pjueon