pjueon/JetsonGPIO #21: error when run simple_out or simple_pwm

edge-ai4cv, 2021-08-11:
Hi

I've just followed your instruction to make and install the lib and compiled the two sample programs. I got the same error message when ran them as:

[Exception] map::at (catched from: get_data())
terminate called after throwing an instance of 'bool'
Aborted (core dumped)

I cannot figure out why this happened and please give me some advice.

Thank you.

edge-ai4cv, 2021-08-12:
Hi,

I've checked out tag v0.1.3. The "simple_out.cpp" seemed fully functional without error!

pjueon, 2021-08-17:
Hi, thank you for reporting.
Did the error happen when you use the latest commit of the master branch?

edge-ai4cv, 2021-08-17:
Hi,

Yes I first checked out the latest master branch-31aa91555cca2dd4aeb4022830548010316e75f7 and failed to ran. Then I switched to v0.1.3 and it worked.

pjueon, 2021-10-13:
Sorry for the late update. I've been too busy to debug.
I've tested on Jetson Nano 2GB model, but could not reproduce the error. 

@edge-ai4cv Can you give us more details about your device please? (model, the version of jetson image, etc)
It would be a huge help for debugging. Thanks.

pjueon, 2021-10-13:
Memo: 
commit 738afcad for the master branch might be the reason (not sure).