pjueon/JetsonGPIO #26: Jetpack 4.6 -- error: ‘enable_if_t’ in namespace ‘std’ does not name a template type        template <class T, class = std::enable_if_t<!std::is_same<std::decay_t<T>, Callback>::value>

afridali123, 2021-11-19:
Getting this below after upgrading to jetpack 4.6

/usr/local/include/JetsonGPIO.h:164:39: error: ‘enable_if_t’ in namespace ‘std’ does not name a template type
       template <class T, class = std::enable_if_t<!std::is_same<std::decay_t<T>, Callback>::value>>
                                       ^~~~~~~~~~~
/usr/local/include/JetsonGPIO.h:164:50: error: expected ‘>’ before ‘<’ token
       template <class T, class = std::enable_if_t<!std::is_same<std::decay_t<T>, Callback>::value>>
                                                  ^
/usr/local/include/JetsonGPIO.h: In lambda function:
/usr/local/include/JetsonGPIO.h:167:83: error: ‘decay_t’ is not a member of ‘std’
         comparer([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                                   ^~~~~~~
/usr/local/include/JetsonGPIO.h:167:83: note: suggested alternative: ‘decay’
         comparer([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                                   ^~~~~~~
                                                                                   decay
/usr/local/include/JetsonGPIO.h:167:83: error: ‘decay_t’ is not a member of ‘std’
/usr/local/include/JetsonGPIO.h:167:83: note: suggested alternative: ‘decay’
         comparer([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                                   ^~~~~~~
                                                                                   decay
/usr/local/include/JetsonGPIO.h:167:64: error: parse error in template argument list
         comparer([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/JetsonGPIO.h: At global scope:
/usr/local/include/JetsonGPIO.h:210:54: error: could not convert ‘nullptr’ from ‘std::nullptr_t’ to ‘const GPIO::Callback&’
                         unsigned long bounce_time = 0);
                                                      ^
/usr/local/include/JetsonGPIO.h:211:115: error: could not convert ‘nullptr’ from ‘std::nullptr_t’ to ‘const GPIO::Callback&’
    void add_event_detect(int channel, Edge edge, const Callback& callback = nullptr, unsigned long bounce_time = 0);
                                                                                                                   ^
app_plunger/CMakeFiles/motor.dir/build.make:62: recipe for target 'app_plunger/CMakeFiles/motor.dir/src/motor.cpp.o' failed
make[2]: *** [app_plunger/CMakeFiles/motor.dir/src/motor.cpp.o] Error 1
CMakeFiles/Makefile2:984: recipe for target 'app_plunger/CMakeFiles/motor.dir/all' failed
make[1]: *** [app_plunger/CMakeFiles/motor.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/afrid/Desktop/release_v2.1/release_v2.1/piece_level/src/app_plunger/src/sw_trigger.cpp:6:0:
/usr/local/include/JetsonGPIO.h:164:39: error: ‘enable_if_t’ in namespace ‘std’ does not name a template type
       template <class T, class = std::enable_if_t<!std::is_same<std::decay_t<T>, Callback>::value>>
                                       ^~~~~~~~~~~
/usr/local/include/JetsonGPIO.h:164:50: error: expected ‘>’ before ‘<’ token
       template <class T, class = std::enable_if_t<!std::is_same<std::decay_t<T>, Callback>::value>>
                                                  ^
/usr/local/include/JetsonGPIO.h: In lambda function:
/usr/local/include/JetsonGPIO.h:167:83: error: ‘decay_t’ is not a member of ‘std’
 er([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                     ^~~~~~~
/usr/local/include/JetsonGPIO.h:167:83: note: suggested alternative: ‘decay’
 er([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                     ^~~~~~~
                                                                                   decay
/usr/local/include/JetsonGPIO.h:167:83: error: ‘decay_t’ is not a member of ‘std’
/usr/local/include/JetsonGPIO.h:167:83: note: suggested alternative: ‘decay’
 er([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                     ^~~~~~~
                                                                                   decay
/usr/local/include/JetsonGPIO.h:167:64: error: parse error in template argument list
         comparer([](const func_t& A, const func_t& B) { return comparer_impl<std::decay_t<T>>(A, B); })
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/JetsonGPIO.h: At global scope:
/usr/local/include/JetsonGPIO.h:210:54: error: could not convert ‘nullptr’ from ‘std::nullptr_t’ to ‘const GPIO::Callback&’
                         unsigned long bounce_time = 0);
                                                      ^
/usr/local/include/JetsonGPIO.h:211:115: error: could not convert ‘nullptr’ from ‘std::nullptr_t’ to ‘const GPIO::Callback&’
  Edge edge, const Callback& callback = nullptr, unsigned long bounce_time = 0);
                                                                              ^
app_plunger/CMakeFiles/Trigger.dir/build.make:62: recipe for target 'app_plunger/CMakeFiles/Trigger.dir/src/sw_trigger.cpp.o' failed
make[2]: *** [app_plunger/CMakeFiles/Trigger.dir/src/sw_trigger.cpp.o] Error 1
CMakeFiles/Makefile2:1341: recipe for target 'app_plunger/CMakeFiles/Trigger.dir/all' failed
make[1]: *** [app_plunger/CMakeFiles/Trigger.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

afridali123, 2021-11-19:
please help me in fixing this issue @pjueon

pjueon, 2021-11-19:
To compile this library, your compiler must support C++14.

If you're using g++, can you tell me your g++ version?

`g++ --version`

pjueon, 2021-11-20:
Or can you add following code before you include JetsonGPIO and try to compile again ?

```cpp
#if (__cplusplus < 201402L) 
// if C++14 is not supported,

// define std::decay_t, std::enable_if_t
namespace std 
{
	template< class T >
	using decay_t = typename decay<T>::type;

	template< bool B, class T = void >
	using enable_if_t = typename enable_if<B,T>::type;
}

#endif

#include <JetsonGPIO>   // include JetsonGPIO header

```

closed 2021-11-25 by pjueon