pjueon/JetsonGPIO #27: Can't access PWM object in a Custom Class

DanielGM279, 2021-12-01:
Hello, I'm trying to do a custom class in which I will manage several pins of the Jetson Nano board and I came acrross a problem when trying to add a PWM object as a data member of the class, specifying in advance the channel and the frequency it will operate as. 
However, when i try to use the start method the following error pops up:

"Invalid use of member function 'GPIO::PWM myclass::p(int, int)'(did you forget the '()'?)"

Normal output and input pins instantiated from inside a class work perfectly, as I tested with another class. Please find the code below.

```
#include <JetsonGPIO.h>

using namespace GPIO;

class myClass
{
private:
    int channel;
    int frequency;
    PWM p (int channel = 32, int freq = 60);
public:
    myClass(int new_channel = 32, int new_freq = 50);
    void setupPWM();
    void startPWM( double dutyCycle);
};

myClass::myClass( int new_channel, int new_freq)
{
    channel = new_channel;
    frequency = new_freq;
}

void myClass::setupPWM()
{
    setwarnings(false);
    semode(BOARD);
    setup(channel, OUT, LOW);
}

void myClass::startPWM(double dutyCycle)
{
    p.start(dutyCycle);
}
```
I truly do not know what I'm doing wrong, as I need to be able to access all different pins from inside the same class. I don't know if there's a better way to do this, but I'd appreciate the help.

pjueon, 2021-12-01:
Your declaration of myClass::p is worng. You declared it as function that returns PWM type.
You can't  declare a data member like that in c++.

It should be something like this:  
  
```cpp
class myClass
{
private:
    int channel;
    int frequency;
    PWM p;           // declaration of p
public:
    myClass(int new_channel = 32, int new_freq = 50);
    void setupPWM();
    void startPWM( double dutyCycle);
};

myClass::myClass( int new_channel, int new_freq)
 : p(new_channel, new_freq), channel (new_channel),  frequency(new_freq)     // initialize data members here
{
   // this part will be executed after members are initilalized...
}
```

DanielGM279, 2021-12-02:
Thank you for your answer. The error disappeared. However, when wiring the selected pin to an LED it won'r turn on. In the case I follow the same setup process for a PWM object outside of the class the LED does turn on.

Any idea what might be the issue?

pjueon, 2021-12-02:
Maybe it's because you call setupPWM function after p initialized. You should configure the pin first and then create the PWM object.

  
Try something like this:
```cpp
#include<memory>

class myClass
{

private:
    int channel;
    int frequency;
    std::unique_ptr<PWM> p; 
public:
    myClass(int new_channel = 32, int new_freq = 50);
    void setupPWM();
    void startPWM( double dutyCycle);
};

myClass::myClass(int new_channel, int new_freq)
 : p(),   // initialize with nullptr first
channel(new_channel), frequency(new_freq)  
{
   // this part will be executed after members are initilalized...

    setupPWM();.  // configure the pin first
    p = std::make_unique<PWM>(channel, frequency); // create PWM object later.

}

void myClass::startPWM(double dutyCycle)
{
    p->start(dutyCycle);
}
```

pjueon, 2021-12-08:
Close this issue because there is no response.

closed 2021-12-08 by pjueon