pjueon/JetsonGPIO #28: Throw an exception when running wait_for_edge() in a new thread

frostcza, 2021-12-07:
Hello, I'm trying to use GPIO::wait_for_edge() to calculate the high level duration of a PWM signal input like this:

`    struct timeval start, end;

    int mode_pin = 11;
    GPIO::setmode(GPIO::BOARD);
    GPIO::setup(mode_pin, GPIO::IN);
    int high_voltage_time;

    while (!end_this_program) 
    {
        GPIO::wait_for_edge(mode_pin, GPIO::Edge::RISING);
        gettimeofday(&start,NULL);
        GPIO::wait_for_edge(mode_pin, GPIO::Edge::FALLING);
        gettimeofday(&end,NULL);
        high_voltage_time = end.tv_usec - start.tv_usec;
        printf("%d\n",high_voltage_time);
    }`

It works well when executed independently. But I want to check two channels at the same time. And when I create two threads to run the same function, it works well for a while (about 1 minute) and then throws the exception: 

[Exception] Failure to open the /sys/class/gpio/gpio{$ch}/edge file (catched from: GPIO::wait_for_edge())
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failure to open the /sys/class/gpio/gpio{$ch}/edge file
Aborted (core dumped)

Do you have any clue for this? Thank you.

pjueon, 2021-12-08:
@ShimmyShaman Can you check this issue please?

ShimmyShaman, 2021-12-08:
As long as the two channels are _different_ channels the usage is okay. There should've been more error detail (at least to the console) for why the edge file cannot be opened (submitted pull request addresses this).

Try again with that, and at least find out why.

If not then I won't get my hands on my Jetson device for another 2-3 days to check it out.

frostcza, 2021-12-09:
Thanks for the reply. I added the std::perror() and tried again. 
I got an extra message: 

sysfs/edge open: Too many open files

And I checked the /proc/$pid/fd. There are hundreds of records like: 

/sys/devices/2200000.gpio/gpiochip0/gpio/gpio428/value
and
/sys/devices/2200000.gpio/gpiochip0/gpio/gpio445/value

frostcza, 2021-12-09:
I found it is irrelevant to multi-threading. If I run it independently on a single channel for a long time, I will get the same error. 
The complete code is as below. 

`
#include <chrono>
#include <thread>
#include <signal.h>
#include <sys/time.h>
#include <JetsonGPIO.h>

static bool end_this_program = false;
void signalHandler(int s) { end_this_program = true; }
void delay(int s) { std::this_thread::sleep_for(std::chrono::seconds(s)); }

int main()
{
    signal(SIGINT, signalHandler);
    int pwm_input = 11; 
    GPIO::setmode(GPIO::BOARD);
    GPIO::setup(pwm_input, GPIO::IN);
    struct timeval start, end;
    int high;

    while (!end_this_program) 
    {
        printf("start to listen...\n");
        GPIO::wait_for_edge(pwm_input, GPIO::Edge::RISING);
        gettimeofday(&start,NULL);
        GPIO::wait_for_edge(pwm_input, GPIO::Edge::FALLING);
        gettimeofday(&end,NULL);
        high = end.tv_usec - start.tv_usec;
        printf("high voltage time: %d us\n", high);
        if(high>1500 && high<=2000)
        {
            printf("received save flag\n");
        }
    }

    GPIO::cleanup();
    return 0;
}
`

ShimmyShaman, 2021-12-12:
Yeap, you're right, a file wasn't being closed when it should've been. Good error report.

Added pull request. The files get closed now and the records don't pile up in /proc/$pid/fd

frostcza, 2021-12-12:
Thanks! Now it runs fine.

pjueon, 2021-12-12:
@frostcza  Thank you for confirming and the error report.
@ShimmyShaman Thank you for your help.

Merged the pull request. Closed the issue.

closed 2021-12-12 by pjueon