pjueon/JetsonGPIO #32: pca9684 in jetson nano i have done everything possible but this won't control a servo, i did everything from hacks to circuit-python library  still wont work downloaded developer package still nothing went back to jet pack still nothing upgrade from 2g to the original boT jet-son nano all i get is module n

zeaggle, 2021-12-23:
not supported pls help me someone help me i have bout different servos still nothing i tried all the codes carefully i need help

zeaggle, 2021-12-23:
now i ordered the same 16 channel from adafruit some one pls help me i have trued all possible tutorials just to move a servo all i get is error downloaded the git clone repositories nothing adafrui servo nothing circuitpython-pca9685 still nothing hacks nothing toptech.com nothing what is going on pls help me  why sell jetson to the masses?????

ShimmyShaman, 2021-12-24:
JetsonHacks did an article & video on this with the pca9685:
https://www.jetsonhacks.com/2019/07/22/jetson-nano-using-i2c/
and Paul McWhorter:
https://www.youtube.com/watch?v=8YKAtpPSEOk

I imagine you've seen that.

You'll have to be more specific with your errors and less verbose with your titles. This isn't really a general help forum, more for issues with the JetsonGPIO library.

Document exactly what you do, and the first place it goes wrong. Then try to solve that.

zeaggle, 2021-12-24:
I am sorry about my approach Sir, and yes i have seen those videos and done
everything step by step and still couldnt move the servo, it was saying
error and the last error i got in the jet-son nano was module not
supported, pls sir if you can help me i will be very grateful

ShimmyShaman, 2021-12-24:
I can say that neither of them would be using this repository, this library, which is a c++ port of the official library.

It may be you got directed here when you should have used the official NVIDIA GPIO python library?
https://github.com/NVIDIA/jetson-gpio

zeaggle, 2021-12-24:
Thanks, I'll check them out. Once i get to my desk, thank you so much Sir

zeaggle, 2021-12-30:
i solved the problem re force the blinka library and connect 5v from your jetson

ShimmyShaman, 2021-12-31:
Good.

closed 2021-12-31 by ShimmyShaman