pjueon/JetsonGPIO #34: Error while trying to include JetsonGPIO Library with find_package(JetsonGPIO REQUIRED)

bart1259, 2022-01-05:
When following the instructions in the readme, I get the following error. I cannot find the required JetsonGPIOConfig.cmake or jetsongpio-config.cmake files anywhere on my system.

    By not providing "FindJetsonGPIO.cmake" in CMAKE_MODULE_PATH this project
      has asked CMake to find a package configuration file provided by
      "JetsonGPIO", but CMake did not find one.
  
      Could not find a package configuration file provided by "JetsonGPIO" with
      any of the following names:
    
        JetsonGPIOConfig.cmake
        jetsongpio-config.cmake
    
      Add the installation prefix of "JetsonGPIO" to CMAKE_PREFIX_PATH or set
      "JetsonGPIO_DIR" to a directory containing one of the above files.  If
      "JetsonGPIO" provides a separate development package or SDK, be sure it has
      been installed.

pjueon, 2022-01-05:
@knro Can you check this issue please?

knro, 2022-01-05:
I am assuming you are using CMake for your project, correct? You need to copy [FindJetsonGPIO.cmake](https://github.com/pjueon/JetsonGPIO/blob/master/cmake_modules/FindJetsonGPIO.cmake) file to your project folder and place it under cmake_modules directory.

pjueon, 2022-01-05:
@knro 
I tested to build a program with CMake according to your instruction, and **I got the same error.** 

My test project
```
test
├── CMakeLists.txt
├── build
├── cmake_modules
│   └── FindJetsonGPIO.cmake
└── foo.cpp
```

CMakeLists.txt _(not working)_
```
cmake_minimum_required(VERSION 3.2)
project(Test LANGUAGES CXX)

find_package(JetsonGPIO REQUIRED)
find_package(Threads REQUIRED)
include_directories(${JetsonGPIO_INCLUDE_DIR})

add_executable(foo foo.cpp)
target_link_libraries(foo ${JetsonGPIO_LIBRARIES} Threads::Threads)
```

If I add `set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake_modules/")` before `find_package(JetsonGPIO REQUIRED)`, it works.

CMakeLists.txt _(working)_
```
cmake_minimum_required(VERSION 3.2)
project(Test LANGUAGES CXX)

set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake_modules/")
find_package(JetsonGPIO REQUIRED)
find_package(Threads REQUIRED)
include_directories(${JetsonGPIO_INCLUDE_DIR})

add_executable(foo foo.cpp)
target_link_libraries(foo ${JetsonGPIO_LIBRARIES} Threads::Threads)
```

Is this supposed to be normal? Or am I missing something?

knro, 2022-01-05:
Right, set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake_modules/") is needed as well. We just use this standard template in our project so I didn't pay attention to this. It's best to update the README to include a sample 3rd party cmake as you did above.

pjueon, 2022-01-05:
Okay, thank you for the confirmation.  I'll update the README soon.

pjueon, 2022-01-05:
Alright, I've just updated README. 

@bart1259  
Could you tell us if this fixes the error?

bart1259, 2022-01-06:
@pjueon the steps in the updated readme worked! Thank you so much!

closed 2022-01-06 by bart1259