pjueon/JetsonGPIO #36: An error while using JetsonGPIO(C++) on Tx2

kidxhac, 2022-01-08:
I  followed the instruction and installed JetsonGPIO v1.0.3, however, when I ran the sample_out.cpp, there occured( other samples failed either):

### terminate called after throwing an instance of  'std::runtime_error'
### what(): [Exception] [model_data]None is not in gpio_chip_dirs (catched from: GPIO::get_data())
### Aborted (core dumped)

Then I used JetsonGPIO in python (sample_out.py), it worked well and pin could output. Nothing changed between two conditions. Needs a little help, thanks.

edge-ai4cv, 2022-01-08:
I have the same issue when I ran your "jetsongpio_test". The hardware is also Tx2 with Jetpack 4.6

pjueon, 2022-01-09:
@kidxhac @edge-ai4cv 
Can you pull the `test_issue36`  branch  of this repository to your local repository and check if this fixes the issue?

edge-ai4cv, 2022-01-09:
@pjueon 

It worked!  Thank you for your prompt fix!

pjueon, 2022-01-09:
@edge-ai4cv Thanks for the confirmation. 

I just merge the branch to the master.

closed 2022-01-09 by pjueon

kidxhac, 2022-01-10:
Sorry for the time difference. I pulled the newest master branch, it worked. Thank you.