pjueon/JetsonGPIO #44: I'm getting `CMake Error at CMakeLists.txt:4 (FetchContent_Declare):   Unknown CMake command "FetchContent_Declare".`

MotorCityCobra, 2022-01-27:
`CMake Error at CMakeLists.txt:3 (include):                                                                                include could not find load file:                                                                                                                                                                                                                 FetchContent                                                                                                                                                                                                                                                                                                                                                        CMake Error at CMakeLists.txt:4 (FetchContent_Declare):                                                                   Unknown CMake command "FetchContent_Declare".                                                                                                                                                                                                                                                                                                                         -- Configuring incomplete, errors occurred!  `

This is from your README where it says "Without installation
If you don't want to install the library you can add it as an external project with CMake's FetchContent:"

pjueon, 2022-01-27:
Thanks for the report. 

@rotolof 
Could you check this issue please?

It might be cmake version issue. 
Could you tell us your cmake version? You can check the version by `cmake --version`

MotorCityCobra, 2022-01-28:
`cmake version 3.10.2`

pjueon, 2022-01-28:
According to the [document](https://cmake.org/cmake/help/latest/module/FetchContent.html), FetchContent is a new feature in 3.11.

So you need to update your cmake. 

I'll add this to README.md later.

rotolof, 2022-01-28:
@pjueon is correct, [FetchContent](https://cmake.org/cmake/help/latest/module/FetchContent.html) was added in 3.11.
You have two options:
- Update CMake
- Use [ExternalProject](https://cmake.org/cmake/help/latest/module/ExternalProject.html) instead. Its syntax is similar.

closed 2022-01-28 by MotorCityCobra