pjueon/JetsonGPIO #45: What is a `target`?

MotorCityCobra, 2022-01-27:
Sorry for the noob question, but what is the "target". I had a CMakeLists.txt that was building fine without a target and now I guess I need one.  

"assuming you added a target called mytarget, then you can link it with:"

`target_link_libraries(mytarget JetsonGPIO::JetsonGPIO)`  

I tried making the "target" the name of my project and the name of the executable. No dice.

MotorCityCobra, 2022-01-27:
I put `target_link_libraries(mytarget JetsonGPIO::JetsonGPIO)` at the last line of my CMakeLists.txt with 'mytarget' the name of my executable and it worked.

closed 2022-01-27 by MotorCityCobra

pjueon, 2022-01-27:
According to [this](https://pabloariasal.github.io/2018/02/19/its-time-to-do-cmake-right/)
> Targets model the components of you application. An executable is a target, a library is a target. Your application is built as a collection of targets that depend on and use each other.

Yes, your executable, or library name is the target