pjueon/JetsonGPIO #46: In the PWM sample file, how would you change the direction?

MotorCityCobra, 2022-01-28:
The script makes my motor go clockwise. How would I make it go counterclockwise?  

```

#include <iostream>
// for delay function.
#include <chrono>
#include <map>
#include <string>
#include <thread>

// for signal handling
#include <signal.h>

#include <JetsonGPIO.h>

using namespace std;

inline void delay(int s) { this_thread::sleep_for(chrono::seconds(s)); }

static bool end_this_program = false;

void signalHandler(int s) { end_this_program = true; }

int main()
{
    // Pin Definitions
    int output_pin = 18;

    // When CTRL+C pressed, signalHandler will be called
    signal(SIGINT, signalHandler);

    // Pin Setup.
    // Board pin-numbering scheme
    GPIO::setmode(GPIO::BOARD);

    // set pin as an output pin with optional initial state of HIGH
    GPIO::setup(output_pin, GPIO::OUT, GPIO::HIGH);
    GPIO::PWM p(output_pin, 200);
    auto val = 75.0;
    p.start(val);

    cout << "PWM running. Press CTRL+C to exit." << endl;
    for (int i = 0; i < 4; i++){
//         if (end_this_program)
        delay(1);
        cout << 1;
    }

    p.stop();
    GPIO::cleanup();

    return 0;
}

```

pjueon, 2022-01-28:
The library only generates PWM signal. 
How your motor moves completely depends on the motor you're using.

First check your motor(what kind of motor, model, etc) and check the data sheet of it.

MotorCityCobra, 2022-01-28:
I thought it might be something to do with the 'high' and 'low'.  
I'm pretty sure it is a software option to change the direction of the motor. On a Raspberry Pi and Arduino IDE I can change the direction with code.

pjueon, 2022-01-28:
There's no direction in pwm signal. Just frequency and duty cycle. 
https://www.analogictips.com/pulse-width-modulation-pwm/

Can you give more details?

What Jetson device are you using?
What kind of motor are you using? (Servo, dc, step, etc)
Motor model?
How do you setup the circuit?

MotorCityCobra, 2022-01-28:
You're probably the most responsive repo owner on GitHub.  

Nvidia Jetson AGX Xavier.  
Nema 17 motor.  

I was just thinking about how I might need to change something with the driver's wiring.

MotorCityCobra, 2022-01-28:
Okay, I got the motor running clockwise and then counterclockwise with the code below.  
I also wired the driver differently. For the signal pins on the driver, all the + sockets are connected to each other and connected to my Jetson's 5V output.  
The signal's dir - is connected to my pin 15.  
The signal's pul - is connected to my pin 18.  

```

#include <iostream>
// for delay function.
#include <chrono>
#include <map>
#include <string>
#include <thread>

// for signal handling
#include <signal.h>

#include <JetsonGPIO.h>

using namespace std;

inline void delay(int s) { this_thread::sleep_for(chrono::seconds(s)); }

static bool end_this_program = false;

void signalHandler(int s) { end_this_program = true; }

int main()
{
    // Pin Definitions
    int output_pin = 18;
    int direction_pin = 15;

    // When CTRL+C pressed, signalHandler will be called
    signal(SIGINT, signalHandler);

    // Pin Setup.
    // Board pin-numbering scheme
    GPIO::setmode(GPIO::BOARD);

    // set pin as an output pin with optional initial state of HIGH

    GPIO::setup(output_pin, GPIO::OUT, GPIO::HIGH);
    GPIO::PWM p(output_pin, 200); 
    GPIO::setup(direction_pin, GPIO::OUT, GPIO::LOW);
    GPIO::PWM e(direction_pin, 200);

    auto val = 75.0;
    e.start(val);
    p.start(val);

    cout << "PWM running. Press CTRL+C to exit." << endl;
    for (int i = 0; i < 4; i++){
        delay(1);
        cout << 1;
    }
    e.stop();
    
    for (int i = 0; i < 4; i++){
        delay(1);
        cout << 1;
    }

    
    p.stop();

    GPIO::cleanup();

    return 0;
}

```

closed 2022-01-28 by MotorCityCobra