pjueon/JetsonGPIO #47: How to prevent PWM-Destructor stopping my PWM after the program ends?

timscheuermann, 2022-01-31:
I try to write a wrapper-function in Simulink with S-Function Builder to get PWM functionality in Simulink for the Jetson Nano.
So I run p.start(DutyCycle[0]) periodically to set new frequency/dutycycle changes for every simulink step, but this leads to flickering LEDs.

I think this happens because when every Simulink step ends, the destructor of the PWM Class stops the PWM signal and the next Simulink step starts it again -> so pwm gets on and off and i have this ugly flickering.

Has someone an idea what i could do or could we build in a parameter to prevent the PWM stop?
Thank you very much.
Greetings, Tim.

pjueon, 2022-01-31:
I'm not a matlab person so I don't really know what is S-Function Builder. 
So I googled it. I guess it's something like exported C++ functions to matlab code, right?

Stopping pwm from the Destructor of `GPIO::PWM` class is the design from the original python library(https://github.com/NVIDIA/jetson-gpio). We want preserve the original design as far as we can.

Here's my suggestion:
Why don't you just keep th PWM object and NOT destruct the PWM object? 

something like this:

Header file
```cpp
extern void init();
extern void start(double dutyCycle);
extern void cleanup();
```

CPP file
```cpp
static GPIO::PWM* pwm = nullptr; 
static bool is_started = false;

void init()
{
    if (pwm == nullptr)
    {
         // initialization routine here...

         // constrcut PWM object
         pwm = new GPIO::PWM(/* .....  */)
        // std::cout << "[DEBUG] pwm object constrcuted" << std::endl;
    }
}

void start(double dutyCycle)
{
    if (pwm == nullptr)
    {
         init();
    }

    if(is_started)
    {
        pwm->ChangeDutyCycle(dutyCycle);
    }
    else
    {
        pwm->start(dutyCycle);
        is_started = true;
    }
}

void cleanup()
{
    if (pwm == nulptr)
        return;
   
    is_started = false;
    delete pwm;
    pwm = nullptr;
    // std::cout << "[DEBUG] pwm object destructed" << std::endl;

    // other cleanup routine here...
}
```

timscheuermann, 2022-01-31:
Thank you for your quick response.
This works like I hoped it would,
The trick with the nullptr solved my problem, 
Thank you very much!

closed 2022-01-31 by timscheuermann