pjueon/JetsonGPIO #51: Cannot use add_event_detect

DanielGM279, 2022-02-10:
Hello, I'm trying to use the add_event_detect with a callback function defined as follows:

`void cb_fcn(const std::string& channel){
    cout << "Inside callback" << endl;
}

my_class::my_class(int new_pinPWMA, int new_Freq)
    :p(createPWM(new_pinPWMA, new_Freq))
{
    p.start(0);
    pulse = 0;
    add_event_detect(pinEncA,RISING, cb_fcn);
}`

However I get the following error when I run the code:

> terminate called after throwing an instance of 'std::runtime_error'
  what():  [Exception] Channel 127 is invalid (catched from: GPIO::add_event_detect())

I followed the interrupt sample but can get my implementation to work, I hope you can help me.

Thank you for your time.

pjueon, 2022-02-10:
Could you tell me which device you're using and the value of pinEncA?

DanielGM279, 2022-02-10:
Hello, I solved the problem already. Turns out the issue was with the order in which I was instantiating the object, which initiated before assigning a value to the pin number, resulting in a trash value.

Thanks for the support.

closed 2022-02-10 by DanielGM279