pjueon/JetsonGPIO #57: Small issues

ShimmyShaman, 2022-03-14:
The requirements for the library have increased with recent commits and implementations of the callback model. I've had to install gcc version 7.5 and add the following to my projects CMakeLists file (c++11 is no longer sufficient).
`add_compile_options(-std=c++14)`
I'm not quite sure what of that is worth mentioning (if at all) in the README.

---
_also... CallbackTypeSelctor is a typo mistake in latest commit_

pjueon, 2022-03-15:
Thank you for the report. 

> I'm not quite sure what of that is worth mentioning (if at all) in the README.

Yeah, I think the C++ standard requirements should be mentioned in README. 
Honestly I thought it was already mentioned in README. I'll fix it soon. 

> _also... CallbackTypeSelctor is a typo mistake in latest commit_

Oh, I didn't noticed it haha. I'll fix it too. Thanks.

pjueon, 2022-03-15:
Ok, I just fixed the issue. 
- Now it can be compiled with C++11 
- Add the C++ standard requirements in the README file

closed 2022-03-15 by pjueon