pjueon/JetsonGPIO #61: Bug when you set a pin as output twice

ma-ludw, 2022-05-09:
If you set a pin as an output twice, the output stops working.

This does not work:
```cpp
  GPIO::setmode(GPIO::BOARD);
  GPIO::setup(32, GPIO::OUT, 0);
  GPIO::setup(32, GPIO::OUT, 0);
  std::this_thread::sleep_for(std::chrono::milliseconds(100));
  GPIO::output(32, 1);
```

I think the problem is, that the pins are exported every time the setup function is called.
I did not test it with input or pwm pins, or switching from input to output.

I use some other library that calls the `GPIO` library and in there, the pin needs the possibility to change the direction.

ma-ludw, 2022-05-10:
As a workaround, the pin needs to be cleaned up manually before every reassignment.

pjueon, 2022-05-16:
Thanks for the report. Fixed it by cleaning up before setup if the channel is already set.

closed 2022-05-16 by pjueon