pjueon/JetsonGPIO #62: [FEA] Improved runtime exception handling

yassiezar, 2022-05-11:
We're working on a fairly complex Jetson-based robotics project and this library forms an important part of the GPIO controller. I understand that your first priority is of course to properly support compilation and running on Jetson platforms. However, its often easier and more convenient to develop and test the software on desktop and deploy it on the robots' Jetsons.

The problem is that when I run software where the GPIO library is used on my desktop, I get a `std::runtime_error` exception seemingly generated when the GPIO singleton is instantiated (the culprit seems to be [this](https://github.com/pjueon/JetsonGPIO/blob/cc504dcf039854d46554319ba5669be13c2d2c35/src/JetsonGPIO.cpp#L136) line in my case). I'm not able to catch this exception, however, since this exception is generated and uncaught inside the GPIO API. If my client was able to catch and handle the exception, I could simply then disable the GPIO handler and eliminate all interactions with it.

My question is if there's a way to make the exception propagate out to the client where it can be handled? Alternatively, how can the code be modified to handle the exceptions internally and make the GPIO API unavailable to clients? I've tried modifying the library code to try and handle the exceptions internally, but I'm running into trouble with the `_cleaner` that runs at the end of the GPIO's lifetime and I can't seem to think of an easy workaround.

pjueon, 2022-05-16:
Ok, now the library will throw an exception on the first API call so you can catch it.

closed 2022-05-16 by pjueon

yassiezar, 2022-05-16:
Thanks for addressing this so quickly @pjueon. I'll try it out and let you know if I run into any issues

pjueon, 2022-05-22:
Improved the API for `GPIO::model` and `GPIO::JETSON_INFO`. 
Now you can use `GPIO::model` and `GPIO::JETSON_INFO`. 
You can also use `GPIO::model()` and `GPIO::JETSON_INFO()` if you want.