pjueon/JetsonGPIO #68: Change PWM phase?

btilmon, 2022-06-18:
Hi. Thanks for making this library. Have you tried implementing a function to adaptively change PWM phase? I am thinking this function will be similar to `ChangeDutyCycle()`. I found a [blog post](https://blog.thea.codes/phase-shifted-pwm-on-samd/) that does this for SAMD21 boards but am not sure how well it translates to the Jetson NANO. I will open a pull request if I figure it out.

pjueon, 2022-06-20:
Hi. I'm not sure if it is even possible to shift phase on the jetson devices. But contribution is always welcome!

pjueon, 2022-08-07:
Close this issue for now as I don't think we'll be able to find a solution to it any time soon. 
Reopen it when it's necessary.

closed 2022-08-07 by pjueon