pjueon/JetsonGPIO #78: Improve detect events skips

automech-rb, 2022-09-20:
Hi,

Thanks for this awesome C++ implementation. 
I am using the `GPIO::add_event_detect(channel, GPIO::BOTH, callback_fn)` feature to detect edge in Jetson AGX Xavier. Environment is JP5.0.2 with docker.

I have a rotary encoder whose digital output change every 15 microseconds from high to low (and vise versa).

The problem is when CPU load increases on Jetson to 50% or more, the `add_event_detect` is not able to pick up all the encoder output change. It don't detect around 30% of the edge (less callbacks) for same encoder RPM. I understand that this is not a RTOS based system, but do you have any recommendations to improve the GPIO callback performance?

thanks

automech-rb, 2022-09-20:
To add, this error behaviour is easy to simulate when comparing output if 15W and MAXN mode. 

For now to mitigate this issue, I am using different thread and looping at 1000hz to check digital input.

pjueon, 2022-09-26:
Thanks for the report. 
 
@ShimmyShaman 
Do you have any idea about this issue?

ShimmyShaman, 2022-09-26:
Distracted by other life things, but I have been looking into it. Believe there are fixes that can be made, but have to test/explore them.

pjueon, 2022-09-27:
@automech-rb 
Does this issue also occur when CPU load is low?

automech-rb, 2022-09-27:
From my testing, this issue only occurs with high CPU usage (over 50%). 
To add more details: 
- I am using two `add_event_detect` for two encoders running inside docker container on JP5.0.2
- Similar error behavior is observed when compute performance is limited by changing mode to 15W (+ more erratic callback sometimes low sometimes high in same timespan for constant encoder RPM)