pjueon/JetsonGPIO #80: Build as a static lib

livneros, 2022-12-18:
Hey,
Is it possible to enable building the library as a static library, rather than a shared one?
Thanks.

livneros, 2022-12-18:
replace:
`ADD_LIBRARY(JetsonGPIO SHARED ${JetsonGPIO_LIB_SRCS})`
with:
`ADD_LIBRARY(JetsonGPIO STATIC${JetsonGPIO_LIB_SRCS})`

closed 2022-12-18 by livneros

KTBM, 2023-12-17:
hello, i am trying to add this library to my project but i can't build a cmake file that's able to make it static.
and i don't understand where to add the line that you've typed and if its a standalone line, because even if I add it, I still receive an error:  attempted static link of dynamic object `/usr/local/lib/libJetsonGPIO.so.1.2.4'

can you please add a CMakeLists.txt file example to make JetsonGPIO a static library.