pjueon/JetsonGPIO #83: Support for Nano and Xavier NX

caruofc, 2023-02-21:
I am having issues with Jetson Nano and Xavier NX. I am using Jetpack 4.6 for both. In Nano, there is no "/proc/device-tree/chosen/plugin-manager/ids" or "/proc/device-tree/chosen/ids" in my system and therefore find_pmgr_board() call in GPIOPinData.cpp fails. And for Xavier NX, none of the matches() call in get_model() returns true and therefore it says "Could not determine Jetson model". So, my question is I guess, does this code support Jetson Nano and Xavier NX? What are the models that it specifically supports? Is there a list of supported models? Thanks.

pjueon, 2023-02-26:
This library does support Nano and Xavier NX. I've only directly tested it on Nano because that's the only one I have, but it should work on Xavier NX because this library is basically just a C++ translation of [Nvidia's official python library](https://github.com/NVIDIA/jetson-gpio).

```
In Nano, there is no "/proc/device-tree/chosen/plugin-manager/ids" or "/proc/device-tree/chosen/ids"
```

That's weird. Are you using customized image or something? I'm using the official Jetson Nano developer kit image and it works. 

Hmm... OK, here's my suggestion. 
Go to [the official python library's repository](https://github.com/NVIDIA/jetson-gpio) and try the library on your devices.
If those paths don't exist, the official python one should not work neither.  

If it doesn't work on your devices, I think it's the best idea to ask them how to fix the issue. Like I said, this library is a C++ translation so it should work same as the original one. If the official python library works on your devices, then we must do something wrong.

pjueon, 2023-10-31:
No response. Closing it.

closed 2023-10-31 by pjueon