pjueon/JetsonGPIO #84: Error compiling with CmakeLists

k1000142, 2023-03-16:
Hi Team, 
Im getting this error when adding your library to our solution. 
We are getting this, any idea what could be?

/usr/local/include/JetsonGPIO/PublicEnums.h: At global scope:
/usr/local/include/JetsonGPIO/PublicEnums.h:64:5: error: expected identifier before ‘,’ token
     PUBLIC_ENUM_CLASS(Directions, UNKNOWN, OUT, IN, HARD_PWM);
/usr/local/include/JetsonGPIO/PublicEnums.h:64:5: error: expected identifier before ‘,’ token
     PUBLIC_ENUM_CLASS(Directions, UNKNOWN, OUT, IN, HARD_PWM);
     ^
/usr/local/include/JetsonGPIO/PublicEnums.h:79:29: error: expected unqualified-id before ‘=’ token
     constexpr Directions IN = Directions::IN;
                             ^
/usr/local/include/JetsonGPIO/PublicEnums.h:80:30: error: expected unqualified-id before ‘=’ token
     constexpr Directions OUT = Directions::OUT;

Thanks

pjueon, 2023-04-03:
Sorry for the late response. I'll check this when I got home.

pjueon, 2023-04-04:
Have you checked the c++ standard version of your project? To build your code with JetsonGPIO, C++11 or higher is required.

pjueon, 2023-05-16:
I added a compilation error message for cases where C++11 is not supported (9a4adf4). 
If you see this message when you re-compile your code with the latest commit, you need  to compile your code with C++11 or a later C++ standard.