pjueon/JetsonGPIO #85: "When importing both JetsonGPIO.h and cuda_runtime.h at the same time, a compilation error occurs."

CLMonster, 2023-04-13:
Hello, 
Due to project needs, I encountered the following error during compilation when I simultaneously imported JetsonGPIO.h and cuda_runtime.h. The code is as follows. 

```cu
#include <cuda_runtime.h>
#include <JetsonGPIO.h>

int main()
{
    return 0;
}
```
[image]
How can I solve this? Thank you very much!

pjueon, 2023-05-11:
Sorry for the late reply. I tested your code on my jetson nano, and I could reproduced the issue. 
Interestingly, when I changed the source file to ".cpp", it compiles perfectly.
```shell
nvcc -o cuda_test cuda_test.cpp -lJetsonGPIO -lpthread
```

I assume that it's some nvcc's name mangling issue but I'm not sure. For now you can make some wrappers for JetsonGPIO in .cpp files and do not include the "JetsonGPIO.h" header file in your ".cu" files. Maybe this could solve the issue.

pjueon, 2023-05-11:
Ah ha! I found it. [This part](https://github.com/pjueon/JetsonGPIO/blob/2a7a2f7a8e84f4e3a9f3e91796fd4226c4491f13/include/JetsonGPIO/TypeTraits.h#L36-L49) of the `TypeTraits.h` was the problem.

I'll fix this soon.

pjueon, 2023-05-11:
Ok, I've just merged the PR and I've confirmed that it fixed the issue. Closing the issue.

CLMonster, 2023-05-14:
> Ok, I've just merged the PR and I've confirmed that it fixed the issue. Closing the issue.

Thank you for helping me. I really appreciate it.